HON95/prometheus-nut-exporter #3: Can’t connect to nut server

bjw-s, 2020-08-25:
Hi,

I’m trying to get this exporter going, but I think I’m doing something wrong in the target. When I just put in the IP and port the exporter can’t connect 😞 I am trying to connect to an instance of https://hub.docker.com/r/instantlinux/nut-upsd/. 

Maybe it’s something in those settings, But they seem pretty vanilla. Maybe you have an example on how to set up a proper NUT server?

Thanks in advance!

HON95, 2020-08-25:
I haven't tried that image, but you should be able to telnet the NUT/UPSD port (3493) and write `LIST UPS` to check connectivity and that it doesn't require authentication. For the exporter you can open `<exporter>/nut?target=<nut>` to directly check its status in case you've messed up the Prometheus config. Note that the target port number is required.

If you want an example on setting up a NUT server you can check my notes: https://haavard.wiki/config/server/linux-applications/#nut

bjw-s, 2020-08-25:
Thanks so much for your notes! That made me get it up and running!

closed 2020-08-25 by bjw-s

dstewen, 2022-04-15:
Unable to access the link to your notes page on how to setup nut. Can you provide an example?