HON95/prometheus-nut-exporter #4: Unable to parse more than one UPS

nvtkaszpir, 2021-03-14:
Looks like the code returns error when trying to process more than one UPS from nut-server.

Example config on the nut server:
ups.conf
```text
[bagien]
    driver = apcsmart
    port = /dev/ttyS0
    desc = "Directly attached to Bagno"

[apcups01]
    driver = snmp-ups
    port = 192.168.1.9
    pollfreq = 10
    community = public
    snmp_version = v2c
    desc = "APC UPS 1K under the desk"
```

running query to prometheus-nut-exporter results in error such as (HTTP 503):
```text
Example error:
```text
Failed to communicate with target.

Malformed list for NUT query "LIST VAR bagien

".
```
and when hitting refresh the word `bagien` will be replaced also with another ups name (in that case `apcups01`).
Also notice that the `LIST VAR <ups>` command actually contains extra newlines, this is weird.

**Things tested:**
- commenting out any part of the code to leave only one UPS - then it works
- testing from another host with single ups - works,
- executed some commands via telnet to investigate, see below (notcie that word `MASKED` was used to hide sensitive data):

```text
VER
Network UPS Tools upsd 2.7.4 - http://www.networkupstools.org/

LIST UPS
BEGIN LIST UPS
UPS bagien "Directly attached to Bagno"
UPS apcups01 "APC UPS 1K under the desk"
END LIST UPS
LIST VAR bagien
BEGIN LIST VAR bagien
VAR bagien battery.alarm.threshold "0"
VAR bagien battery.charge "100.0"
VAR bagien battery.charge.restart "00"
VAR bagien battery.date "01/09/15"
VAR bagien battery.runtime "3300"
VAR bagien battery.runtime.low "120"
VAR bagien battery.voltage "13.85"
VAR bagien battery.voltage.nominal "012"
VAR bagien device.mfr "APC"
VAR bagien device.model "Smart-UPS 620   "
VAR bagien device.serial "MASKED"
VAR bagien device.type "ups"
VAR bagien driver.name "apcsmart"
VAR bagien driver.parameter.pollinterval "2"
VAR bagien driver.parameter.port "/dev/ttyS0"
VAR bagien driver.parameter.synchronous "no"
VAR bagien driver.version "2.7.4"
VAR bagien driver.version.internal "3.1"
VAR bagien input.frequency "50.00"
VAR bagien input.quality "FF"
VAR bagien input.sensitivity "H"
VAR bagien input.transfer.high "253"
VAR bagien input.transfer.low "208"
VAR bagien input.transfer.reason "unacceptable utility voltage rate of change"
VAR bagien input.voltage "230.4"
VAR bagien input.voltage.maximum "230.4"
VAR bagien input.voltage.minimum "228.9"
VAR bagien output.voltage "230.4"
VAR bagien output.voltage.nominal "230"
VAR bagien ups.delay.shutdown "020"
VAR bagien ups.delay.start "000"
VAR bagien ups.firmware "22.6.I"
VAR bagien ups.id "BAGIEN  "
VAR bagien ups.load "014.9"
VAR bagien ups.mfr "APC"
VAR bagien ups.mfr.date "09/20/01"
VAR bagien ups.model "Smart-UPS 620   "
VAR bagien ups.serial "MASKED"
VAR bagien ups.status "OL"
VAR bagien ups.test.interval "1209600"
VAR bagien ups.test.result "NO"
END LIST VAR bagien
LIST VAR apcups01
BEGIN LIST VAR apcups01
VAR apcups01 ambient.1.humidity.alarm.high "60.00"
VAR apcups01 ambient.1.humidity.alarm.low "30.00"
VAR apcups01 ambient.1.temperature.alarm.high "40.00"
VAR apcups01 ambient.1.temperature.alarm.low "10.00"
VAR apcups01 battery.charge "100.00"
VAR apcups01 battery.charge.restart "15"
VAR apcups01 battery.current "0.00"
VAR apcups01 battery.date "10/06/15"
VAR apcups01 battery.packs "0.00"
VAR apcups01 battery.packs.bad "0.00"
VAR apcups01 battery.runtime "9180.00"
VAR apcups01 battery.runtime.low "120"
VAR apcups01 battery.voltage "27.50"
VAR apcups01 battery.voltage.nominal "0.00"
VAR apcups01 device.mfr "APC"
VAR apcups01 device.model "Smart-UPS 1000"
VAR apcups01 device.serial "MASKED"
VAR apcups01 device.type "ups"
VAR apcups01 driver.name "snmp-ups"
VAR apcups01 driver.parameter.pollfreq "10"
VAR apcups01 driver.parameter.pollinterval "2"
VAR apcups01 driver.parameter.port "192.168.1.9"
VAR apcups01 driver.parameter.snmp_version "v2c"
VAR apcups01 driver.parameter.synchronous "no"
VAR apcups01 driver.version "2.7.4"
VAR apcups01 driver.version.data "apcc MIB 1.2"
VAR apcups01 driver.version.internal "0.97"
VAR apcups01 input.frequency "50.00"
VAR apcups01 input.sensitivity "medium"
VAR apcups01 input.transfer.high "253"
VAR apcups01 input.transfer.low "208"
VAR apcups01 input.transfer.reason "selfTest"
VAR apcups01 input.voltage "234.70"
VAR apcups01 input.voltage.maximum "236.10"
VAR apcups01 input.voltage.minimum "233.20"
VAR apcups01 output.current "0.00"
VAR apcups01 output.frequency "50.00"
VAR apcups01 output.voltage "234.70"
VAR apcups01 output.voltage.nominal "220"
VAR apcups01 ups.delay.shutdown "90"
VAR apcups01 ups.delay.start "0"
VAR apcups01 ups.firmware "600.3.I"
VAR apcups01 ups.id "UPS_1K"
VAR apcups01 ups.load "5.80"
VAR apcups01 ups.mfr "APC"
VAR apcups01 ups.mfr.date "04/23/05"
VAR apcups01 ups.model "Smart-UPS 1000"
VAR apcups01 ups.serial "MASKED"
VAR apcups01 ups.status "OL"
VAR apcups01 ups.temperature "28.80"
VAR apcups01 ups.test.date "03/11/2021"
VAR apcups01 ups.test.result "Ok"
END LIST VAR apcups01

HON95, 2021-03-21:
Thanks for reporting this! It seems like something very simple to fix, but I just haven't found the time yet.

HON95, 2021-04-07:
This should be fixed now (in #8). The fix is available in Docker tag `latest` but is not yet released  (in tag `1` etc.). I realize you're probably not interested any more since this didn't work at all for you, but if you happen to try it I would be interested to hear if it works for you now. Thanks for all the provided info!

closed 2021-04-07 by HON95

nvtkaszpir, 2021-04-07:
Just tested, it's working fine :)
Thank you very much!

nvtkaszpir, 2021-04-07:
[image]

heh even got a power outage today ;D

nvtkaszpir, 2021-04-16:
Just wanted to say that version docker.io/hon95/prometheus-nut-exporter@sha256:f381d4bca35696cc7497942bd1ed225b02208154dd5562ff6a8b6998480b5ec5 works witout any restarts for over 9 days.