HON95/prometheus-nut-exporter #5: Add power value

Roken120, 2021-04-06:
Hello,
Can you add this field :
ups.power
ups.power.nominal
ups.model
input.frequency
input.voltage.nominal
output.frequency
output.frequency.nominal

HON95, 2021-04-06:
See the comment in #6. I can't accept the PR as-is as it has errors and some potential for improvement, but I can fix it if you give me the changes by opening a new PR with the same changes to a temporary branch like `2021-04-07/new-metrics`, where I can continue to work on it. Battery type and device model is already added as labels for the `nut_device_info` metric as they're text and not numerical values.

Roken120, 2021-04-06:
Hello
Yes you can change and push no problem.

HON95, 2021-04-07:
Since this (that) is your PR, I don't have acces to change the target. So do you wish to open a new PR targeting the temporary branch i created (just copy-paste the contents of common.rs, I suppose), or should I just copy the contents of the file in your PR and fix it myself, then close/reject your PR? Up to you.

Also, would you be able to provide a text dump of your NUT/UPS vars so I have something to test against? (`telnet <nut-ip> <nut-port>`, then run `LIST UPS` and `LIST VAR <ups-name>`.) Alternatively I can just use the examples on the NUT website if getting your vars is too much trouble.

Roken120, 2021-04-07:
battery.charge: 97
battery.charge.low: 20
battery.charge.warning: 20
battery.runtime: 1693
battery.temperature: 29.9
battery.type: PbAc
battery.voltage: 40.1
battery.voltage.nominal: 40.0
device.mfr: Hewlett-Packard
device.model: HP T1500 G3  UPS
device.part: DF321A
device.serial: 3C848550538
device.type: ups
input.frequency: 50.0
input.transfer.high: 266.0
input.transfer.low: 194.0
input.voltage: 230.0
input.voltage.nominal: 230
output.current: 1.3
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 100
ups.load: 31
ups.load.nominal: 80
ups.mfr: Hewlett-Packard
ups.model: HP T1500 G3  UPS
ups.power: 302.0
ups.power.nominal: 1400
ups.productid: 4de5
ups.serial: 3C82477538
ups.status: OL CHRG
ups.test.result: Done and passed
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.timer.start: -1
ups.vendorid: 03f0

HON95, 2021-04-07:
Thanks for the vars! (I don't need the serial if you wish to hide it.)

HON95, 2021-04-07:
I think I've added all the metrics you asked for now and some more, could you have a look? For the vars you posted above, it looks like [this](https://gist.github.com/HON95/5cecc60482b1d42201bc420135ad7929).

closed 2021-04-07 by Roken120