HON95/prometheus-nut-exporter #10: Can't connect to NUT server running on synology

Shamazo, 2021-04-10:
I apologize if the issue ends up not being in this repo. 

I have a NUT server running on my synology. I _think_ I can telnet to it correctly:
```
[hamish@hamish-desktop home-infra]$ telnet nut.nicholson.systems 3493                                                                                                                  
Trying 10.0.0.11...                                                                                                                                                                           
Connected to nut.nicholson.systems.                                                                                                                                                           
Escape character is '^]'.                                                                                                                                                                     
LIST UPS                                                                                                                                                                                      
BEGIN LIST UPS                                                                                                                                                                                
UPS ups "Description unavailable"                                                                                                                                                             
END LIST UPS                                                                                                                                                                                  
^C^]                                                                                                                                                                                          
telnet> exit                                                                                                                                                                                  
?Invalid command                                                                                                                                                                              
telnet> quit                                                                                                                                                                                  
Connection closed.
```

My ups name is creatively ups. When I go to `http://<ip of exporter>:9995/nut?target=nut.nicholson.systems:3493` I get:
```
Failed to communicate with target.

Failed get NUT version from NUT query.
```
The wiki post linked on a very similar issue a few months ago no longer works. Do you have any suggested debugging steps?

HON95, 2021-04-10:
Try running `VER` and share the result. Maybe `LIST VAR ups` as well, but no need to share it if it looks right. From the error message it looks like it successfully connects but fails to read/parse the version.

Shamazo, 2021-04-10:
```
VER
Network UPS Tools upsd DSM6-2-25510-201118 - http://www.networkupstools.org/
LIST VAR ups
BEGIN LIST VAR ups
VAR ups battery.charge "100"
VAR ups battery.charge.low "10"
VAR ups battery.charge.warning "20"
 .. and so on with the rest of the variables 
```

I have a suspicion that that version is slightly non standard because it is packaged by synology.

HON95, 2021-04-10:
I suspect you're right, thanks for reporting this. I'll look into it.

HON95, 2021-04-11:
Could you try the current `latest` image? I'll push a release if it works.

Shamazo, 2021-04-11:
Thank you so much. `latest` is working for me now.

closed 2021-04-11 by HON95