HON95/prometheus-nut-exporter #29: Add "BOOST" to UPS status metrics

nxtByte, 2022-08-02:
Hey,
I just wanted to ask if it's possible for you to add the "BOOST" status to the metrics. The "BOOST" status is activated, when the input voltage is too low (in my area <215 Volts) and the UPS needs to "boost" the Voltage to match the expected output (in my area 220-240 Volts). As I quite often get an "undervoltage" here and the UPS's need to boost, I'd like to add it to my monitoring, so I can provide the data to my electricity provider, so maybe they can fix it. Would it be possible for you to add that to the metrics?
Thanks in advance!
[image]

HON95, 2022-08-02:
Sure, could you send the text in the screenshot as plaintext so have something to test against?

nxtByte, 2022-08-02:
Thanks for the quick reply!
Of course, here you go:
`root@raspberrypi:~# upsc greencell@localhost
Init SSL without certificate database
battery.charge: 100
battery.runtime: 2667
battery.voltage: 13.60
battery.voltage.high: 13.80
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.runtimecal: 400,100,800,50
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.internal: 0.12
input.current.nominal: 8.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 209.3
input.voltage.fault: 209.3
input.voltage.nominal: 230
output.voltage: 246.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 15
ups.productid: 0000
ups.status: OL BOOST
ups.type: offline / line interactive
ups.vendorid: 0001`

closed 2022-08-03 by HON95

HON95, 2022-08-03:
v1.2.1 should be available from Docker Hub in about 10 minutes.

nxtByte, 2022-08-03:
Thank you so much! This was really nice and quick! :)

PR #30 (2022-08-03): Add BOOST status
Fixes #29.

## Changes

- Added `BOOST` status to the `nut_ups_status` state set.