HON95/prometheus-nut-exporter #38: Can't get it working

adrianmihalko, 2024-11-23:
I am trying to get this working, but for some reason it's not working for me.

When I try to access: http://10.88.1.24:9995/nut?target=10.88.1.25:3493 I see Not found error.

My docker config is:

```
version: '3.3'
volumes:
  prometheus-data:
    driver: local
  grafana-data:
    driver: local
services:
  prometheus:
    image: prom/prometheus:latest
    container_name: prometheus
    restart: unless-stopped
    volumes:
      - ./config:/etc/prometheus/
      - prometheus-data:/prometheus
    networks:
      - prometheus-network
    ports:
      - "9090:9090"
  grafana:
    image: grafana/grafana:latest
    container_name: grafana
    restart: unless-stopped
    volumes:
      - grafana-data:/var/lib/grafana
    networks:
      - prometheus-network
    ports:
      - "3000:3000"
  node_exporter:
    image: quay.io/prometheus/node-exporter:latest
    container_name: node_exporter
    command:
      - '--path.rootfs=/host'
    pid: host
    ports:
      - "9100:9100"
    restart: unless-stopped
    volumes:
      - '/:/host:ro,rslave'
    networks:
      - prometheus-network
  nut-exporter:
    # Stable v1
    image: hon95/prometheus-nut-exporter:1
    environment:
      - TZ=Europe/Budapest
      - HTTP_PATH=/metrics
      # Defaults
      #- RUST_LOG=info
      #- HTTP_PORT=9995
      #- HTTP_PATH=/nut
      #- LOG_REQUESTS_CONSOLE=false
      #- PRINT_METRICS_AND_EXIT=false
    ports:
      - "9995:9995/tcp"
    networks:
      - prometheus-network      
networks:
  prometheus-network:
     driver: bridge 
```

My config/prometheus.yml:

```
# my global config
global:
  scrape_interval: 60s  # Global scrape interval
  scrape_timeout: 10s   # Global scrape timeout

  # Attach these labels to any time series or alerts when communicating with
  # external systems (federation, remote storage, Alertmanager).
  external_labels:
    monitor: 'codelab-monitor'

# A scrape configuration containing exactly one endpoint to scrape:
# Here it's Prometheus itself.
scrape_configs:
  - job_name: 'prometheus'
    static_configs:
      - targets: ['prometheus:9090']
  - job_name: 'node_exporter'
    static_configs:
      - targets: ['node_exporter:9100']
  - job_name: 'ubuntu_1.23'
    static_configs:
      - targets: ['10.88.1.23:9100']
  - job_name: 'unifipoller'
    scrape_interval: 30s
    static_configs:
      - targets: ['unpoller:9130']
  - job_name: 'pihole'
    static_configs:
      - targets: ['pihole_exporter:9617']
  - job_name: 'nut'
    static_configs:
      # Insert NUT server address here
      - targets: ["10.88.1.25:3493"]
    metrics_path: /nut
    relabel_configs:
      - source_labels: [__address__]
        target_label: __param_target
      - source_labels: [__param_target]
        target_label: instance
      - target_label: __address__
        # Insert NUT exporter address here
        replacement: 10.88.1.24:9995

```

On 10.88.1.25 3493 I tried to telnet and I saw data.

What I am doing wrong?

GuideGlyph, 2025-03-24:
Hi

I had the same problem. After some debugging i notice

```yaml
# docker compose
# ...
  nut-exporter:
    image: hon95/prometheus-nut-exporter:1
    environment:
      - HTTP_PATH=/metrics
# ...
```
you have `HTTP_PATH=/metrics`

```yaml
# prometheus.yaml
# ...
  - job_name: 'nut'
    static_configs:
      # Insert NUT server address here
      - targets: ["10.88.1.25:3493"]
    metrics_path: /nut
# ...
```
You have `metrics_path: /nut`

Just change `/nut/` to `/metrics`

HON95, 2025-03-27:
Nice catch, I've updates the README.

closed 2025-03-27 by HON95